LibreSolar/mppt-2420-hc #1: Some questions

moritz31, 2020-11-30:
I have a few questions:

Could this Board be soldered by hand or do i need to Order it.

What is the minimal Input for this Board ? I currently have a 3W Solar Panel, will That work ?

martinjaeger, 2020-11-30:
It can still be hand-soldered if you have some soldering experience with fine pin-pitch parts. I usually solder the first prototypes by hand aswell.

3W is most probably too small and the control algorithms would most probably not work properly anymore. This board is more suited for something up to 300W (at a 12V battery).

moritz31, 2020-12-03:
Thank you for the answer. I have done different electrical stuff already and soldered some smaller chips so it is Worth a try.

For the power side: i am currently building a Battery pack at 24V and the Solar side is not determined, is there something which works with a 20-50W Solar Panel ?

theonaldovncts, 2023-07-16:
Hey I want to ask, If I want to adopt the charge controller for a wind turbine, which terminal do I need to connect to dump load?