LibreSolar/mppt-2420-hc #2: Inductor saturation for 24V batteries

martinjaeger, 2021-04-14:
The inductor layout turned out to be too optimistic for full power (20A) in 24V systems. As the ripple current increases with the output voltage, the inductor is driven into saturation and the core gets too warm (>80°C).

We are currently doing further testing with a larger inductor wound around two stacked cores such that it will still fit to the PCB as it is now. Updates after testing will be posted here.

intrinsicaly, 2023-11-19:
Did you mange to progress any further with this one?

martinjaeger, 2023-11-20:
Yes, kind of. It works very well with larger inductor, but the board design would have to be updated to fit the inductor.

intrinsicaly, 2023-11-26:
Is that something you are considering by any chance?

martinjaeger, 2023-11-27:
Not at the moment, unfortunately. Have to focus on some paid-for projects first.

intrinsicaly, 2023-11-30:
I appreciate that. Thanks for your works thus far.

intrinsicaly, 2024-03-08:
Hi Martin,
I note on your schematic pdf that you have 2 stacked ms-130060-2 cores. Is that in response to this issue of saturation or is this the original design?

martinjaeger, 2024-03-08:
Two stacked cores is one possible solution. Another solution would be a larger core in general.
The original design had only one core.

intrinsicaly, 2024-03-11:
Thanks for that.
I appreciate your time while you are busy with paid work. 
Would you be able to help this novice with some info from the design.
Looking at the design schematic I can see that you are targeting 53uH for the inductor, 21 turns 27AWG 40 strand copper wire but I am unsure of the info below;
VLon (V) and VLoff (V) and f(Hz)

glenndk, 2025-06-18:
> The inductor layout turned out to be too optimistic for full power (20A) in 24V systems. As the ripple current increases with the output voltage, the inductor is driven into saturation and the core [...]

Please look at:
https://github.com/LibreSolar/mppt-2420-hc/issues/10