LibreSolar/mppt-2420-hc #3: 160V from PV possible or more?

dPeS, 2021-05-31:
first of all - great project!

context: I'm mostly IT guy - familiar with coding but also electronics is my hobby. I have 4 x 450W PV (41V shorted) in series that I use for heating water via MPPT controler (commercial). Durring sunny day at 13:00-14:00 water is already at 65C and want to redirect that 160V (4x40) into 4x12V acid accu.

Is it possible to change that mosfet into more powerfull and drive 160V or more?

martinjaeger, 2021-05-31:
Unfortunately, that won't be easily possible. It's not only the MOSFET, but also several other components that would need to be changed, e.g. the capacitors and the MOSFET driver. It's basically a new PCB design...

dPeS, 2021-05-31:
@martinjaeger but it is possible to make 12v output from 160V?

if you guide me I could replace driver + buy caps for higher volts - you will have free stress testing :]

martinjaeger, 2021-05-31:
It is possible in general, but the input voltage steps for the MPPT algorithm become very coarse because of the high ratio between input and output voltage. This microcontroller runs at 150 MHz, which is already quite good for the PWM signal resolution.

I don't think there is a pin-compatible driver for higher voltage (with similar drive current capabilities). If you find one, let me know...

dPeS, 2021-06-21:
@martinjaeger IPP110N20NA from infineon will do the job?

X-Ryl669, 2021-11-04:
You can have 2 board and plug 2 PV panels in series on one board and the other 2 on another board. You'll then charge 2 12V batteries independently. Beware not to connect the batteries on the same load circuit (since it would share the different voltage ground).

DodoidOfficial, 2025-04-21:
> I don't think there is a pin-compatible driver for higher voltage (with similar drive current capabilities). If you find one, let me know...

Apologies for the reply to the ancient issue. I am also interested in using the MPPT-2040-HC with higher-voltage panels. The panels I am interested in are ~150Voc each, so unfortunately shortening the string is not an option (and most off-the-shelf MPPTs won't work!). I've already made a few PCB tweaks to accommodate larger capacitors on DCDC_HV+ and increase a few clearances for a 200V rating.

Suitable drop-in FETs and capacitors seem to be available. For the gate driver, the [IRS21867STRPBF](https://www.mouser.ca/datasheet/2/196/Infineon_IRS21867S_DataSheet_v01_00_EN-3363603.pdf) looks like a drop-in replacement. The only downside I can see is longer (up to 250ns) propagation times (I couldn't totally figure out the range of PWM frequencies this uses - are these too long?). Other than that it has 4A of drive current, 3.3V-compatible inputs, it looks to be the same pinout, should run off the 12V supply, and it's rated up to 600V.

Do you see any reason this wouldn't work?

echedey-ls, 2025-05-01:
> > I don't think there is a pin-compatible driver for higher voltage (with similar drive current capabilities). If you find one, let me know...
> 
> [@martinjaeger](https://github.com/martinjaeger) Apologies for the reply to the ancient issue. I am also interested in using the MPPT-2040-HC with higher-voltage panels. The panels I am interested in are ~150Voc each, so unfortunately shortening the string is not an option (and most off-the-shelf MPPTs won't work!). I've already made a few PCB tweaks to accommodate larger capacitors on DCDC_HV+ and increase a few clearances for a 200V rating.
> 
> Suitable drop-in FETs and capacitors seem to be available. For the gate driver, the [IRS21867STRPBF](https://www.mouser.ca/datasheet/2/196/Infineon_IRS21867S_DataSheet_v01_00_EN-3363603.pdf) looks like a drop-in replacement. The only downside I can see is longer (up to 250ns) propagation times (I couldn't totally figure out the range of PWM frequencies this uses - are these too long?). Other than that it has 4A of drive current, 3.3V-compatible inputs, it looks to be the same pinout, should run off the 12V supply, and it's rated up to 600V.
> 
> Do you see any reason this wouldn't work?

@DodoidOfficial I'm interested in this conversation. From [the schematics](https://github.com/LibreSolar/mppt-2420-hc/blob/main/build/mppt-2420-hc_schematic.pdf), I think the main problem could be with the 12V supply circuit, the voltage follower in page 3. It will need to provide more current due to increased gate capacitance. I don't know how much nor if it can handle it well. What MOSFETs were you talking about?

DodoidOfficial, 2025-05-03:
@echedey-ls I was looking at the MASPOWER [MS130N20JDT0](https://www.lcsc.com/product-detail/MOSFETs_MASPOWER-MS130N20JDT0_C37635851.html). You can get similar, more-name-brand parts from Mouser, etc but it looked like you'd be spending $40+ on FETs alone if you did.